WiringPi/WiringPi-Python #17: Upload wheel to PyP

willprice, 2015-04-23:
It would be nice to have a wheel (a precompiled binary installation package) available on PyPi to minimize the time taken to install the package. This can be achieved by 
```shell
$ python setup.py sdist bdist_wheel upload
```
Or with *twine* (as recommended in the [python packaging guide](https://packaging.python.org/en/latest/distributing.html#uploading-your-project-to-pypi))
```shell
$ pip install twine
$ twine upload dist/*
```

Gadgetoid, 2015-04-24:
This is an awesome protip, and I'll look into it as soon as possible. This could come in useful for all the other packages I maintain, too, since a wheel dist should in theory work on both old and new Pi's... for now! I think WiringPi relies on run-time version detection ( rather than the compile time stuff I've used before ) so there shouldn't be anything Pi-version-specific to stop it from just-working.

willprice, 2015-04-24:
That's a good point you make about pi version specific things. You might have to compile a version on both versions to get the armv6 and armv7 versions and upload both of them.

Gadgetoid, 2016-07-22:
Of note, I tried to upload an armv7 wheel recently for a different package and was stonewalled by a message to the tune of "unsupported architecture." I'll have to look into this more.

Gadgetoid, 2017-08-12:
Ben Nuttall has apparently solved this issue now, with his own Wheel build server and custom distribution URL. See: http://bennuttall.com/piwheels-building-a-faster-python-package-repository-for-raspberry-pi-users/

closed 2017-08-12 by Gadgetoid

bennuttall, 2017-11-07:
@Gadgetoid You can now upload ARM wheels to PyPI (the default upload URL has been changed to legacy.python.org and that now accepts ARM wheels). But if you don't, piwheels will build it and provide a wheel that way.