WiringPi/WiringPi-Python #19: Build fails

LaraCraft93, 2015-06-13:
The commit 60cc642 breaks the build for me. I'm using Arch Linux ARM

Here is the fail log:

In file included from wiringpi_wrap.c:2984:0:
WiringPi/wiringPi/mcp23x0817.h:82:0: warning: "IOCON_INIT" redefined
 #define IOCON_INIT (IOCON_SEQOP)
 ^
 ^
wiringpi_wrap.c: In function 'PyInit__wiringpi2':
wiringpi_wrap.c:8087:21: warning: variable 'md' set but not used [-Wunused-but-set-variable]
   PyObject *m, *d, *md;
                     ^
In file included from wiringpi_wrap.c:2998:0:
wiringpi_wrap.c: At top level:
WiringPi/devLib/font.h:15:22: warning: 'font' defined but not used [-Wunused-variable]
 static unsigned char font [] =
                      ^
build/temp.linux-armv6l-3.4/WiringPi/devLib/piFaceOld.o: In function `myDigitalRead':
/home/lara/Develop/aur4/python-wiringpi2-git/src/WiringPi2-Python/WiringPi/devLib/piFaceOld.c:110: multiple definition of `myDigitalRead'
build/temp.linux-armv6l-3.4/WiringPi/devLib/piFace.o:/home/lara/Develop/aur4/python-wiringpi2-git/src/WiringPi2-Python/WiringPi/devLib/piFace.c:59: first defined here
build/temp.linux-armv6l-3.4/WiringPi/devLib/piFaceOld.o: In function `myDigitalWrite':
/home/lara/Develop/aur4/python-wiringpi2-git/src/WiringPi2-Python/WiringPi/devLib/piFaceOld.c:87: multiple definition of `myDigitalWrite'
build/temp.linux-armv6l-3.4/WiringPi/devLib/piFace.o:/home/lara/Develop/aur4/python-wiringpi2-git/src/WiringPi2-Python/WiringPi/devLib/piFace.c:45: first defined here
build/temp.linux-armv6l-3.4/WiringPi/devLib/piFaceOld.o: In function `myPullUpDnControl':
/home/lara/Develop/aur4/python-wiringpi2-git/src/WiringPi2-Python/WiringPi/devLib/piFaceOld.c:134: multiple definition of `myPullUpDnControl'
build/temp.linux-armv6l-3.4/WiringPi/devLib/piFace.o:/home/lara/Develop/aur4/python-wiringpi2-git/src/WiringPi2-Python/WiringPi/devLib/piFace.c:74: first defined here
build/temp.linux-armv6l-3.4/WiringPi/devLib/piFaceOld.o: In function `piFaceSetup':
/home/lara/Develop/aur4/python-wiringpi2-git/src/WiringPi2-Python/WiringPi/devLib/piFaceOld.c:159: multiple definition of `piFaceSetup'
build/temp.linux-armv6l-3.4/WiringPi/devLib/piFace.o:/home/lara/Develop/aur4/python-wiringpi2-git/src/WiringPi2-Python/WiringPi/devLib/piFace.c:90: first defined here
collect2: error: ld returned 1 exit status
error: command 'gcc' failed with exit status 1

Gadgetoid, 2015-06-13:
Aha, this is the piFaceOld.c file tripping it up again. Delete that file and try again, I'll check if there are any erroneous reference to it in the build, since it's not necessary but I don't discern what gets inherited from WiringPi.

LaraCraft93, 2015-06-13:
It's working now, but the file resurfaces after some compilations.

closed 2015-12-13 by ShyPixie