WiringPi/WiringPi-Python #20: lcdCharDef(int handle, int index, unsigned char data [8])

speendo, 2015-07-30:
Yesterday I tried to define a new character on my Hitachi HD44780U 2004 LCD.

When I issue something like

    # Ä
    cap_umlaut_a = [
        int('0b01010', 2),
        int('0b00100', 2),
        int('0b01010', 2),
        int('0b10001', 2),
        int('0b11111', 2),
        int('0b10001', 2),
        int('0b10001', 2),
        int('0b00000', 2)
    ]

    print(cap_umlaut_a) # [10, 4, 10, 17, 31, 17, 17, 0]

    wiringpi.lcdCharDef(lcd_handle, 0, cap_umlaut_a)

I get the following error message:

> TypeError: in method 'lcdCharDef', argument 3 of type 'unsigned char [8]'

In `lcdPutchar (int handle, unsigned char data)` I had a similar error that I could solve with

    wiringpi.lcdPutchar(lcd_handle, ord('A'))

However, I cannot find a solution for the issue I have with `lcdCharDef()`.

Is this a bug?

(Please also note my question on stackoverflow: http://stackoverflow.com/questions/31711978/unsigned-char-array-from-int-array-in-python/31712747)

speendo, 2015-07-30:
Also note https://github.com/WiringPi/WiringPi2-Python/issues/13

Gadgetoid, 2016-02-28:
This should be fixed in: https://github.com/WiringPi/WiringPi-Python/commit/8e29d40b0235ef1efe06e5611b8f8b93e713a4f0

I found your SO question quite by accident when researching how to accomplish this. I took a different approach- allowing a regular Python list as long as it's:

* Exactly 8 items long
* Contains only integers between 0 and 255

I'm sure there are myriad other ways to accomplish this, but not having to fuss about in Python is a bonus!

closed 2017-08-12 by Gadgetoid