WiringPi/WiringPi-Python #21: piBoardRev: Unable to determine board revision from /proc/cpuinfo

marshalls, 2015-08-03:
My python script runs perfectly for about an hour and then errors out with this error: 

piBoardRev: Unable to determine board revision from /proc/cpuinfo

I'm only calling  wiringpi2.wiringPiSetup() once in my script.  

I'm using a Rpi 2.  Any ideas?

sawik-PL, 2016-02-02:
got the same problem!
the reason is multiple call wiringPiSetup which opens /dev/mem, which is never closed; this causes errno=24  /* Too many open files */
i created simple patch for this:
diff --git a/WiringPi/wiringPi/wiringPi.c b/WiringPi/wiringPi/wiringPi.c
index 2382112..4540018 100644
--- a/WiringPi/wiringPi/wiringPi.c
+++ b/WiringPi/wiringPi/wiringPi.c
@@ -647,7 +647,7 @@ int wiringPiFailure (int fatal, const char *message, ...)
 static void piBoardRevOops (const char *why)
 {
   fprintf (stderr, "piBoardRev: Unable to determine board revision from /proc/cpuinfo\n") ;
-  fprintf (stderr, " -> %s\n", why) ;
+  fprintf (stderr, " -> %s errno=%d\n", why,errno) ;
   fprintf (stderr, " ->  You may want to check:\n") ;
   fprintf (stderr, " ->  http://www.raspberrypi.org/phpBB3/viewtopic.php?p=184410#p184410\n") ;
   exit (EXIT_FAILURE) ;
@@ -1819,6 +1819,8 @@ unsigned int micros (void)
 
 int wiringPiSetup (void)
 {
+  static int call_count=0;
+  if (call_count) return 0;
   int   fd ;
   int   boardRev ;
   int   model, rev, mem, maker, overVolted ;
@@ -1905,6 +1907,7 @@ int wiringPiSetup (void)
   else
     wiringPiMode = WPI_MODE_PINS ;
 
+  call_count++;
   return 0 ;
 }

liffiton, 2017-08-16:
This appears to have been fixed in a similar way in a newer revision of WiringPi ([here](https://github.com/WiringPi/WiringPi/blob/c9476436016c995d3327e4765a73de7848d4af56/wiringPi/wiringPi.c#L2142) in the current version).  This library includes that fix now, so I'll close this.

closed 2017-08-16 by liffiton