WiringPi/WiringPi-Python #23: Was not working until refreshed with latest "wiringpi2"

christophe-dupriez, 2016-01-06:
Hi!
I first installed WiringPi2-Python following the instructions in README.md
It was not working (digitalRead was always returning "1").
I deleted the following directories created under WiringPi2-Python:
build, dist, wiringpi2.egg-info, WiringPi
I then installed the latest "wiringPi2" and moved it to the directory "WiringPi2-Python/WiringPi2" (uppercase W)
I re-issued : sudo python setup.py install

Would it be possibile to repackage WiringPi2-Python with latest version of WiringPi2 ?

Thanks !  Christophe

Gadgetoid, 2016-02-16:
Yes, this is something I need to do.

Do you, and anyone else who cares to comment, favour a switch to using a submodule, rather than a hard-copy of WiringPi?

I prefer a hard-copy for simplicity, and to support patching of the codebase should it ever be needed without adding a whole lot of new steps to the build process. But I'm not staunchly opposed to a submodule.

Gadgetoid, 2016-02-27:
I've caved in favour of using a submodule - however it's linked from github.com/wiringPi/wiringPi rather than the original repository. This is so I can manage the removal of things like piFaceOld.c which breaks the build, and which I don't want to try and script around.

I'll endeavor to keep the mirror properly updated!

closed 2016-02-28 by Gadgetoid