WiringPi/WiringPi-Python #25: wiringPiSPIDataRW with one byte payload fails

ihcfan, 2016-02-13:
Under python 2.7.6

The code below fails if the same call is made twice from Python 2.7.6.  I cannot confirm other python versions.
`
    buffer = str(bytearray([value]))
    wiringPiSPIDataRW(0,buffer)
`

It appears that the buffer gets overwritten with the response from the ioctl call, and subsequent calls with the same parameter (passed in as value) result in the erroneous overwritten byte being transferred, rather than the actual value specified.  This problem does not appear if the buffer is longer than 1 character.

obtitus, 2016-04-19:
Here is a minimal example that fails as long as MISO is not shorted to MOSI, demonstrating the behaviour noticed by ihcfan. The behaviour is not very pythonic, as one would expect the string passed to wiringPiSPIDataRW to be immutable.

```python
import wiringpi
wiringpi.wiringPiSetup()

fd = wiringpi.wiringPiSPISetupMode(1, 5000, 0)
if fd == -1:
    raise IOError('Unable to initialize SPI %s' % fd)

hexdata = chr(0xff) + chr(0x55)
assert hexdata == chr(0xff) + chr(0x55) # passes

length, read_data = wiringpi.wiringPiSPIDataRW(1, hexdata)

assert hexdata == chr(0xff) + chr(0x55) # fails, as hexdata is now overwritten with responce.
```

Tested with python 2.7.9. Python does not support copying a string, since you should never need too, but the following is a workaround from http://stackoverflow.com/questions/24804453/how-can-i-copy-a-python-string:
```python
length, read_data = wiringpi.wiringPiSPIDataRW(1, "%s" % hexdata)
```

jsl2, 2016-11-26:
Adding a malloc in the typemap fixes this
```
%typemap(in) (unsigned char *data, int len) {
      $2 = PyString_Size($input);
      $1 = (unsigned char *) malloc($2);
      memcpy($1, (unsigned char *) PyString_AsString($input), $2);
};

%typemap(freearg) (unsigned char *data) {
      free((unsigned char *) $1);
}

```
I would have made a pull request, but I'm not sure why `%typemap(in) (unsigned char *data, int len)` exists twice in wiringpi.i

Gadgetoid, 2017-08-12:
I'm resurrecting this thread since I believe this issue still exists and it is, in fact, not very Pythonic. The suggested solution looks sane enough, but I'm not completely confident it wont have repercussions.

bvwelch, 2018-02-10:
Greetings,

First, having a python wrapper for wiringpi is fantastic!

But I am seeing what I think is this same issue.  Here is a little snippet.  After the first pass thru the loop, the 'A' is changed to zero !

while 1:
    bozo = 'A'
    print 'bozo', hex( ord(bozo) )
    trans = 'A'
    got = wi.wiringPiSPIDataRW(SPIchannel, trans)

If I change the trans assigment, like so:  trans = 'B', then bozo will print out as A. ( 0x41 )

William

liffiton, 2018-02-13:
@bvwelch, yes, that very likely is the same issue.  It's especially surprising to see the function modify a string that isn't even referenced in that line, but I believe that's due to Python interning short strings (aliasing a single object to both bozo and trans, here).

The workaround suggested above should work: force a copy of any string you pass to wiringPiSPIDataRW with string formatting, e.g. ``"%s" % trans``.

liffiton, 2018-04-30:
I got some time to dig into this more, and I have a preferred fix, but it would break most existing code, or there's a potential middle ground...

SWIG supports a [buffer interface](http://www.swig.org/Doc3.0/Python.html#Python_nn75) for Python, and the `%pybuffer_mutable_binary` macro fits very well with the `wiringPiSPIDataRW()` function.  That would let one pass any Python object with a buffer interface (most simply a `bytearray` object) to the function, and the function would modify the contents of that buffer (just as it does in C).

This is a change from the current implementation, which accepts strings in Python 2 and bytes in Python 3.  The lack of consistency there isn't great.  And the current implementation lets one pass it an "immutable" string that the C function then proceeds to modify, which is the root cause of this issue.

So "fixing" the method by more closely matching the C function's interface would mean any code passing it a plain string or bytes object would no longer work, and it would standardize on `bytearray` objects.  It would also avoid all of the issues that lie between strings and bytes, though, like encoding/decoding, differences between Python 2 and 3, etc.

Fixing it by allowing input as it does currently (with the difference between Py2/3) but making a copy of the input first (the solution above using `memcpy()`) feels messier to me, as it leaves open all of those issues, but it would probably work and it would be best for backward compatibility.

Which brings me to an intermediate solution that might work:  Have the function translate any argument (string, byte, or bytearray) into a bytearray first, then pass that to the C function cleanly with `%pybuffer_mutable_binary`, and return that bytearray (as modified by the C function).  This keeps the input flexibility (and even increases it), though the return type does change.  We could possibly translate the bytearray back into a string, but that doesn't feel great: the SPI data may not be ASCII, and the user may want raw bytes anyway.

I'm interested in others' opinions on this.  Any fix will change the behavior in _some_ way.  At the very least, fixing this issue means that anyone who relies on the argument being modified by the function would lose that "feature."  So maybe it's a question of how much breakage is acceptable in return for a "cleaner" fix.

ul-gh, 2019-01-06:
Hi @Gadgetoid and @liffiton, thank you for this code and for the support already.

I am using the library as you can see from above reference. The way I see it, the current status is that there is no consistent way to run the same code without modification in Python 2.7 and Python 3.

Tested stock Raspbian Stretch as of today:

The syntax suggested at https://github.com/WiringPi/WiringPi-Python#quick-install does not work for me in Python 2.7:
`buf = bytes([your data here])`
`retlen, retdata = wiringpi.wiringPiSPIDataRW(0, buf)`

And the workaround with the string argument copied using string formatting syntax causes above segfault with Python 3.

==> If you ask me, I would be glad to accomodate any small change to the existing code in order to have it running without caring about the Python version - and any fix would be better than none..

Regards, Ulrich

liffiton, 2019-01-22:
@ul-gh, I am leaning toward making a code-breaking change to make it more sensible.  However, I'm not sure exactly what issues you are running into here.

The following code works for me in both Python 2 and 3:

```python
import wiringpi

wiringpi.wiringPiSPISetup(0, 500000)
buf = bytes("hello".encode('utf8'))
retlen, retdata = wiringpi.wiringPiSPIDataRW(0, buf)
print(retlen)
print(repr(retdata))
```

```
$ python2 test25.py 
5
'\x00\x00\x00\x00\x00'

$ python3 test25.py 
5
b'\x00\x00\x00\x00\x00'
```

Does something like that work for you?  If so, I could update the README to more clearly convey this.

[Any actual development work will have to wait until my hand is out of this splint, which might be a bit...]

ul-gh, 2019-01-24:
Am 22.01.19 um 23:42 schrieb Mark Liffiton:
> @ul-gh <https://github.com/ul-gh>, I am leaning toward making a
> code-breaking change to make it more sensible. However, I'm not sure
> exactly what issues you are running into here.

Hi, thanks for your answer, indeed apart from the changing of the
supposedly not mutable function argument, the problem is finding a
syntax that works both with the string type for Pyton 2, bytes type for
Python 3, supports bitwise manipulation etc..

I think the struckt Python module has what I need in order to convert
between int and bytes/strings in either case. I will have to look into
this further during the next days.

Also, seeing that Python 2 is reaching end of support next year, some of
this overhead could be avoided using the Python 3 builtin features.

I will do some experimenting and hopefully get my code working soon.

ul-gh, 2019-01-25:
Adding to the above, the newly suggested workaround also does not work in Python 2.x:
```
buf = bytes("hello".encode('utf8'))
retlen, retdata = wiringpi.wiringPiSPIDataRW(0, buf)
```
This fails because in Python 2, bytes() ist just an alias for str() and str() cannot handle unicode characters with ordinal code point values above 127.

I.e. `chr(255).encode("utf-8")` in Python 2 fails with:
`UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 0: ordinal not in range(128)`.

The way I see it, you can only send 7-Bit ASCII characters... 

Finally, I found a workaround that works equally well in Python 2 and Python 3:

```
include struct
def _send_uint8(self, val):
        # Reads an integer, sends its modulo-256 value via the SPI bus as uint-8
        wp.wiringPiSPIDataRW(self.SPI_CHANNEL, struct.pack("B", val&0xFF))
```

This seems like a good solution for larger strings to be transferred, but the strut module has some overhead for one-byte-values AFAIK.

Maybe it is just better to do just completely do without the Python 2 compatibility instead, seeing the overhead all this brings and seeing the 2020 deadline?

closed 2023-10-31 by Gadgetoid