WiringPi/WiringPi-Python #26: Feature Request: support wiringPiSPISetupMode

ihcfan, 2016-02-13:
I have been unable to find a way using WPI2-Python to control clock polarity (CPOL and CPHA flags).  Under wiring pi the function wiringPiSPISetupMode allows a third parameter to specify these.  As an interim fix I simply recompiled the code and hijacked the wiringPiSPISetup wrapper to accept a third parameter and specify the mode.  However, it would be nice if this functionality was exposed in the base.

ihcfan, 2016-02-13:
SWIGINTERN PyObject *_wrap_wiringPiSPISetup(PyObject *SWIGUNUSEDPARM(self), PyObject *args) {
  PyObject *resultobj = 0;
  int arg1 ;
  int arg2 ;
  int arg3 ;
  int val1 ;
  int ecode1 = 0 ;
  int val2 ;
  int ecode2 = 0 ;
  int val3 ;
  int ecode3 = 0 ;
  PyObject * obj0 = 0 ;
  PyObject * obj1 = 0 ;
  PyObject * obj2 = 0 ;
  int result;
  
  if (!PyArg_ParseTuple(args,(char *)"OOO:wiringPiSPISetup",&obj0,&obj1,&obj2)) SWIG_fail;
  ecode1 = SWIG_AsVal_int(obj0, &val1);
  if (!SWIG_IsOK(ecode1)) {
    SWIG_exception_fail(SWIG_ArgError(ecode1), "in method '" "wiringPiSPISetup" "', argument " "1"" of type '" "int""'");
  } 
  arg1 = (int)(val1);
  ecode2 = SWIG_AsVal_int(obj1, &val2);
  
  if (!SWIG_IsOK(ecode2)) {
    SWIG_exception_fail(SWIG_ArgError(ecode2), "in method '" "wiringPiSPISetup" "', argument " "2"" of type '" "int""'");
  } 
  arg2 = (int)(val2);
  
  ecode3 = SWIG_AsVal_int(obj2, &val3);
  if (!SWIG_IsOK(ecode3)) {
    SWIG_exception_fail(SWIG_ArgError(ecode3), "in method '" "wiringPiSPISetup" "', argument " "3"" of type '" "int""'");
  }
  arg3 = (int)(val3);
  
  result = (int)wiringPiSPISetupMode(arg1,arg2,arg3);
  resultobj = SWIG_From_int((int)(result));
  return resultobj;
fail:
  return NULL;
}

Gadgetoid, 2016-02-28:
Should be fixed in: https://github.com/WiringPi/WiringPi-Python/commit/d7bdef9defdd97fff333041ffa46f55a73043810

I'll collect up any unwrapped functions and push a new build to pip.

liffiton, 2017-08-13:
[Cleaning up old issues]
As noted, this should be fixed by that commit.  (Feel free to open a new commit if the fix / that function doesn't work.)

closed 2017-08-13 by liffiton