WiringPi/WiringPi-Python #27: Rename to "WiringPi"?

Gadgetoid, 2016-02-28:
For legacy reasons, now long lost in the mists of time, this repository was originally called "WiringPi2-Python" and the library it generates and which I deploy to pip is still called "WiringPi2".

I'd like to propose a rename to "WiringPi", replacing the now ancient "WiringPi" package on pip with the latest and greatest, and perhaps changing WiringPi2 to a stub package with a wrapper and redirect to WiringPi. ( Or some other, more pip-like method of deprecating and redirecting. )

Before diving in and doing this, I wanted to ask if there's anything I might potentially break? I suspect probably not. But thought it best to ask.

raspitv, 2016-02-28:
I don't use it a lot these days, but when I did I always used to start with...

import WiringPi2 as WiringPi  so my scripts didn't break changing from 1 and 2

I can't really see what it would break to be honest Phil, and WiringPi (1) is about 3 years old now.

Alex

Gadgetoid, 2016-03-02:
It's clear I should just rename it and worry about the consequences later. It would have more short term benefits than drawbacks having a not stupid name.

closed 2017-08-12 by Gadgetoid