WiringPi/WiringPi-Python #29: pinMode() unsupported with the /sys/class/gpio backend

Gadgetoid, 2016-02-28:
Issue copied from: https://github.com/Gadgetoid/WiringPi2-Python/issues/32

@chrisspen wrote:

The function `wiringpi2.pinMode()` appears to do nothing when using the `wiringPiSetupSys()` backend. I have my permissions set to that any user belonging to the gpio group can access the sys files, but I can't access the pins until I export the gpio pins outside of wiringpi2.

The [docs](http://wiringpi.com/reference/setup/) for wiringpi do say you have to separately export the GPIO pins when using the /sys/class/gpio backend...but I don't understand the reason for this.

I would expect wiringpi2 to enable a pin as an output via:

    wiringpi2.pinMode(mypin, 1)

but instead, I have to do:

    os.system('cd /sys/class/gpio; echo {mypin} > export; echo out > gpio{mypin}/direction'.format(mypin=mypin))

Once that's done, the rest of the wiringpi2 interface works as expected.

Am I missing something? Why is `pinMode()` essentially a noop when using the /sys/class/gpio backend, instead of a convenience wrapper around an `os.system()` call?

stevieb9, 2016-08-17:
The reason for this is because `wiringPiSetupSys()` is designed to allow you to run your scripts with non-root users, at the cost of some functionality loss. `pinMode()` touches hardware directly which requires root. Exporting the pins manually bypasses that, which allows non-root operation. 

The same thing goes for `pullUpDnControl()`.

chrisspen, 2016-08-17:
If the user doesn't have access to resources, shouldn't you throw an exception instead of ignoring the call altogether, regardless of whether or not they have access?

It's trivially easy to enable access to gpio for non-root users, as [explained in this question](http://raspberrypi.stackexchange.com/questions/40105/access-gpio-pins-without-root-no-access-to-dev-mem-try-running-as-root), but even when the user has full access, WiringPi still doesn't work.

Gadgetoid, 2017-08-12:
I believe with the `/sys/class/gpio` backend WiringPi doesn't even *try* to access the low-level hardware, since the interface doesn't define any way for accessing pull up/down states. These have to be set directly in the registers.

Indeed, the pinMode() wrapper should probably be customised to raise a Python exception (NotImplementedError?) in cases where pinMode() is likely to do nothing.

stevieb9, 2017-08-12:
Raising `NotImplementedError` is likely the most reasonable approach here, with an explicit warning to the user exactly what isn't implemented and perhaps why. Now, whether to exit after a warning or not is a different story.

Gadgetoid, 2017-08-16:
I'll consider this an enhancement, rather than a bug fix, although it does blur the lines between the two. I can't guarantee there will be a graceful way to trap accesses to pull/up down states while in `/sys/class/gpio` mode in the bindings alone so, given that we avoid making any modifications to the underlying WiringPi library as a matter of course, it may be impossible to implement this.

Still, I'll look into it!

GM-Script-Writer-62850, 2018-09-03:
For me pinMode can be observed in `/sys/class/gpio/gpio7/direction` given you export the pin, this also works using RPi.GPIO
but my script is running as root
It is worth noting that root permission is not required for using the /sys/class/gpio/ on my install of Raspbian stretch

closed 2023-10-31 by Gadgetoid