WiringPi/WiringPi-Python #33: Cannot import in Python 3.4

robertjensen, 2016-03-02:
Hi
Python 3.4 boots many enhancements over Python 3.2 and is also the default version in newer versions of Raspbian. Is there any hope that WiringPi-Python will be updated to allow import under Python 3.4?

Currently the import fails with the following error message:
>>> import wiringpi2 as wp
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.4/dist-packages/wiringpi2-1.1.1-py3.4-linux-armv6l.egg/wiringpi2.py", line 26, in <module>
    _wiringpi2 = swig_import_helper()
  File "/usr/local/lib/python3.4/dist-packages/wiringpi2-1.1.1-py3.4-linux-armv6l.egg/wiringpi2.py", line 22, in swig_import_helper
    _mod = imp.load_module('_wiringpi2', fp, pathname, description)
  File "/usr/lib/python3.4/imp.py", line 243, in load_module
    return load_dynamic(name, filename, file)

Best regards
robert

Gadgetoid, 2016-03-02:
I swear I tested Python 3 support on bleeding edge Raspbian on a Pi 3! I'm also struggling to see the error in this error message. Will look into it next week.

Gadgetoid, 2016-03-02:
This is what I see having built from the GitHub source using SWIG3.0 on a Pi Zero running Raspbian Jessie:

```
pi@raspberrypi:~ $ sudo python3
Python 3.4.2 (default, Oct 19 2014, 13:31:11)
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import wiringpi2 as wpi
```

Have you tried:

```
sudo pip install wiringpi2 --upgrade
```

It looks like the package version you're running is ancient, it should read:

```
wiringpi2-2.32.0-py3.4-linux-armv6l
```

robertjensen, 2016-03-02:
Hi Gadgetoid
Thank you very much for investigating this. I will try you suggestion tomorrow when I get back to the lab.

robertjensen, 2016-03-03:
Good news first: I have now successfully imported wiringpi2 under Python 3 :-)

However, I am very confused regarding the difference between WiringPi-Python and WiringPi2-Python, I think I might have accidentally installed WiringPi2-Python which is properly the reason you noticed the ancient version. I thus tried to install WiringPi-Python which seems to me to be the non-deprecated one, however, I was unable to follow the install instructions. The swig-command gives me the following error:
pi@rasppi72 ~/WiringPi-Python $ swig3.0 -thread -python wiringpi.i
swig error : Unrecognized option -thread
Use 'swig -help' for available options.

And trying to install anyway immediately fails: 
pi@rasppi72 ~/WiringPi-Python $ sudo python3 setup.py install
Traceback (most recent call last):
  File "setup.py", line 10, in <module>
    sources.remove('WiringPi/devLib/piFaceOld.c')
ValueError: list.remove(x): x not in list

So, I went to pip which actually worked. However, as from you suggestion I used the command
sudo pip3 install wiringpi2
Which I would think was the deprecated version?

However, in the end it actually works and I am happy :-)

For completeness I should mention this was tested on Debian 8, Jessie.

Regards
Robert

Gadgetoid, 2016-03-09:
It's a little confusing at the moment. "WiringPi2-Python" has now been renamed to "WiringPi-Python" on GitHub, but is still called "WiringPi2" on PIP. You have correctly installed the latest and greatest version, but reminded me that I really need to tidy that up!

netsky-z, 2016-04-01:
between WiringPi-Python and WiringPi2-Python, I don't know which one is the latest version, I really confusing that, im looking for a long time about this on the internet, but now, aha, I know it ~, thank you so much Gadgetoid.

closed 2016-07-20 by Gadgetoid

fabianWhoknows, 2017-06-11:
Hi there
I had the following problem when following the installation procedure for WiringPi-Python on github:

```
pi@raspberry ~/WiringPi-Python $ swig3.0 -thread -python wiringpi.i
swig error : Unrecognized option -thread
Use 'swig -help' for available options.
```
As it turned out its a simple spelling mistake in the instruction: instead of -thread it should be -thread**s**.
This solved the issue for me