WiringPi/WiringPi-Python #37: wiringpi.wiringPiISR depends on the gpio program

joerick, 2016-06-10:
When installing wiringpi using pip, it all builds and installs fine, but when installed, there's a dependency on the gpio program from WiringPi. e.g.

```python
>>> import wiringpi
>>> wiringpi.wiringPiSetupGpio()
0
>>> wiringpi.pinMode(18, 0)
>>> wiringpi.digitalRead(18)
0
>>> wiringpi.digitalRead(18)
1
>>> from __future__ import print_function
>>> wiringpi.wiringPiISR(18, 1, lambda: print('a'))
wiringPiISR: Can't find gpio program
wiringPiISR: unable to open /sys/class/gpio/gpio18/value: No such file or directory
```

Workaround is to install via apt as well.

    sudo apt-get install wiringpi

Gadgetoid, 2016-07-20:
This seems to be intentional behavior in wiringPi, I'm not sure what the rationale behind it is, but it's partially explained here: 

https://github.com/WiringPi/WiringPi/blob/9a8f8bee5df60061645918231110a7c2e4d3fa6b/wiringPi/wiringPi.c#L1848-L1852

> // Now export the pin and set the right edge
> //	We're going to use the gpio program to do this, so it assumes
> //	a full installation of wiringPi. It's a bit 'clunky', but it
> //	is a way that will work when we're running in "Sys" mode, as
> //	a non-root user. (without sudo)

Might be worth me noting that `apt-get install wiringpi` is a dependency for installation. Cheers.

joerick, 2016-07-20:
Thanks for the response, @Gadgetoid. Hmm. Probably no way around it, especially given the absolute path to the utility in the C code.

TominJoseph18, 2017-01-30:
Hi.

I want to send interrupts through my gpio pins. I am using the wiringPi Library for this purpose. When a code to wait for interrupt is run ,the ethernet connection to my laptop breaks and it says broken pipe error.

Could someone please help me to execute a command X in C when an interrupt event occurs which also prevents any kind of switch debouncing and transient voltages.

Thank you.

PR #39 (2017-08-12): Add `apt-get install wiringpi` to README
Fix #37
commit 1b9c63aa (2017-08-12): Add `apt-get install wiringpi` to README

Fix #37
closed 2017-08-12 by Gadgetoid