WiringPi/WiringPi-Python #38: **Error with build.sh**

GitSujal, 2016-06-27:
I'm getting this error with ./build.sh 

*pi@raspberrypi*:~/WiringPi-Python $ ./build.sh 
Traceback (most recent call last):
  File "setup.py", line 10, in <module>
    sources.remove('WiringPi/devLib/piFaceOld.c')
ValueError: list.remove(x): x not in list
python: can't open file 'test.py': [Errno 2] No such file or directory

Gadgetoid, 2016-07-20:
This is because the contents of the `WiringPi` directory have not yet been populated. It's a Git submodule, so you must:

```
git submodule update --init
```

As far as I can see, I've neglected to mention this on the README. I'll amend that!

liffiton, 2017-08-13:
[Cleaning up old issues.]
The earlier commit fixed this, and build.sh is now unneeded/gone as well.

closed 2017-08-13 by liffiton