WiringPi/WiringPi-Python #40: [Help] installation problem

ShunJianLee, 2016-08-10:
Anyone know how to solve this problem?

pi@raspberrypi:~ $ git clone git://git.drogon.net/wiringPi
Cloning into 'wiringPi'...
warning: expected SRV RR, found RR type 1
fatal: unable to connect to git.drogon.net:
git.drogon.net[0: 195.10.226.169]: errno=Connection timed out
git.drogon.net[1: 2a00:ce0:2:feed:beef:cafe:0:3]: errno=Network is unreachable

Gadgetoid, 2016-08-10:
Gordon's Git server seems a little sluggish today. Either keep trying, or use my GitHub mirror instead ( which is up to date as far as I'm aware ):

```
git clone https://github.com/wiringpi/wiringpi
```

closed 2016-08-10 by Gadgetoid

ykla, 2019-10-28:
Now I also can't git it. And github wiring not updates for a year. So how to download it ?

liffiton, 2019-10-28:
Gordon seems to have shut down his [git server](https://git.drogon.net/):

> git.drogon.net is temporarily unavailable. Please look for alternatives for wiringPi, etc. -Gordon

I'm guessing this is related to his decision to [stop developing WiringPi](http://wiringpi.com/wiringpi-deprecated/).

I'm not sure whether he has released any source code newer than what is mirrored in [WiringPi/WiringPi](https://github.com/WiringPi/WiringPi).

Based on a pull-request in that repository, it seems that a user has made his/her own updates for the RPi 4: [ifel/WiringPi](https://github.com/ifel/WiringPi)

ykla, 2019-10-29:
@liffiton he released 2.5.2 to support raspberry pi 4. But I can't found it.