WiringPi/WiringPi-Python #42: Where to go to remove sudo requirement? Got study and book suggestion?

lmckusic, 2016-08-18:
I apologize for pestering you with issue requests.

The wiringpi for Python requirement for sudo is requiring me to rewrite the shell script or change the permissions of the shell script. I am switching a heater on and off. Forcing every Python3 command to be preceded by sudo I wish to stop.

What file contains the instructions requiring sudo and could you please suggest me a tutorial that might help me figure out the chain of programs and processes that generate the wiringpi.py module?

pikowai, 2016-09-11:
Use the gpio utility provide by wiringpi (http://wiringpi.com/the-gpio-utility/) to set pins up how you want (this may be done in a shell script) and then call wiringpi.wiringPiSetupsys() at the beginning of your python script. If you are using a 4.1 kernel then you will also need to add your user to the gpio group and add a rule as described here https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=118879#

liffiton, 2017-08-13:
Recent version of WiringPi with a recent kernel do not require sudo for most GPIO access.  sudo is still required for using PWM output, but I am not aware of any way around this with WiringPi at the moment.  As an alternative, the [pigpio](http://abyz.co.uk/rpi/pigpio/) library lets you run a long-running daemon process as root, and then user-level (non-sudo) processes can communicate with it to use the GPIO unrestricted.

closed 2017-08-13 by liffiton