WiringPi/WiringPi-Python #44: Install failure

geovogel, 2016-09-25:
Followed instructions, everything going fine during installation then;

arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -fPIC -IWiringPi/wiringPi -IWiringPi/devLib -I/usr/include/python3.4m -c wiringpi_wrap.c -o build/temp.linux-armv7l-3.4/wiringpi_wrap.o
arm-linux-gnueabihf-gcc: error: wiringpi_wrap.c: No such file or directory
arm-linux-gnueabihf-gcc: fatal error: no input files
compilation terminated.
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 4

closed 2016-10-14 by geovogel

frakman1, 2017-06-06:
What is the solution to this problem? I ran into the same error.

ipachip, 2017-06-08:
This? https://discuss.flexget.com/t/solved-flexget-update-and-install-errors/1695/4