WiringPi/WiringPi-Python #46: Unable to install wiringpi to alpine linux which uses musl

2m, 2017-02-21:
Got this error when trying to install wiringpi:

```
[Build]        gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -IWiringPi/wiringPi -IWiringPi/devLib -I/usr/local/include/python3.6m -c WiringPi/wiringPi/wiringPiSPI.c -o build/temp.linux-armv7l-3.6/WiringPi/wiringPi/wiringPiSPI.o
[Build]        In file included from /usr/include/sys/ioctl.h:7:0,
[Build]                         from WiringPi/wiringPi/wiringPiSPI.c:30:
[Build]        WiringPi/wiringPi/wiringPiSPI.c: In function 'wiringPiSPIDataRW':
[Build]        WiringPi/wiringPi/wiringPiSPI.c:89:35: error: '_IOC_SIZEBITS' undeclared (first use in this function)
[Build]           return ioctl (spiFds [channel], SPI_IOC_MESSAGE(1), &spi) ;
[Build]                                           ^
[Build]        WiringPi/wiringPi/wiringPiSPI.c:89:35: note: each undeclared identifier is reported only once for each function it appears in
[Build]        error: command 'gcc' failed with exit status 1
```

Looks like the problem was fixed in the upstream wiringpi: http://lists.busybox.net/pipermail/buildroot/2016-June/163445.html

liffiton, 2017-08-15:
I just tested this in an Alpine VM, and it built fine (after I installed all needed packages).  A regular `pip install wiringpi` completed successfully, and `import wiringpi` succeeded in a python interpreter session.

I'm guessing this was fixed by the recent release that includes the newer WiringPi version.  Can either of you test it and let me know if it works for you?

2m, 2017-08-16:
I have originally ran across this error while building a resin.io image. Now it works. I used `3.6.1` tag of https://hub.docker.com/r/resin/raspberry-pi-alpine-python/tags/ and successfully installed `wiringpi==2.44.4`.

closed 2017-08-16 by 2m

liffiton, 2017-08-16:
Great!  Thanks for verifying.