WiringPi/WiringPi-Python #47: /proc/cpuinfo reports BCM2835 on 4.9 kernel, causing unworking wiringpi

razerraz, 2017-03-08:
First, thanks for your work.

I get one big issue on a fresh raspberry pi2 archlinux arm, coming with 4.9.13 kernel. The /proc/cpuinfo reports now hardware as BCM2835 :  
```
processor	: 0
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 1
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 2
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 3
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

Hardware	: BCM2835
Revision	: a01041
Serial		: 000000004d7f80f7
```

The fix i've found is in wiringPi.c, adding or statement with BCM2835.
If It can be done upstream with no side effects.

Thanks

neuralassembly, 2017-03-28:
I have an identical problem. 

This problem has already been fixed in the latest version (2.44) of the upstream WiringPi.
However, the WiringPi in WiringPi-Python (WiringPi-Python/WiringPi) is very old (2.32).

So I replaced "WiringPi-Python/WiringPi" with the latest one,
and installed WiringPi-Python again.

However, the following ImportError on "crypt" appeared.
I hope that WiringPi-Python works with the latest WiringPi.

```
pi@raspberrypi:~ $ python
Python 2.7.9 (default, Sep 17 2016, 20:26:04)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import wiringpi
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/wiringpi-2.32.1-py2.7-linux-armv7l.egg/wiringpi.py", line 28, in <module>
    _wiringpi = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/wiringpi-2.32.1-py2.7-linux-armv7l.egg/wiringpi.py", line 24, in swig_import_helper
    _mod = imp.load_module('_wiringpi', fp, pathname, description)
ImportError: /usr/local/lib/python2.7/dist-packages/wiringpi-2.32.1-py2.7-linux-armv7l.egg/_wiringpi.so: undefined symbol: crypt
```

neuralassembly, 2017-03-28:
I made a small progress. I could run the WiringPi-Python with the latest WiringPi
on kernel 4.9.

The root privilege is still required.
If you omit "sudo", the Rapsberry Pi will stall.

[How to install]
```
$ git clone https://github.com/WiringPi/WiringPi-Python.git
$ cd WiringPi-Python
$ rm -rf WiringPi
$ git clone https://github.com/WiringPi/WiringPi.git
```

Then please apply the following patch to WiringPi-Python/setup.py

```
--- setup.py.orig	2017-03-28 19:53:55.210068570 +0900
+++ setup.py	2017-03-28 19:53:58.190070005 +0900
@@ -12,12 +12,13 @@
 _wiringpi = Extension(
     '_wiringpi',
     include_dirs=['WiringPi/wiringPi','WiringPi/devLib'],
-    sources=sources
+    sources=sources,
+    extra_link_args=['-lcrypt', '-lrt']
 )
 
 setup(
     name = 'wiringpi',
-    version = '2.32.1',
+    version = '2.44',
     author = "Philip Howard",
     author_email = "phil@gadgetoid.com",
     url = 'https://github.com/WiringPi/WiringPi-Python/',
```

Finally, please execute the following commands.

```
$ cd WiringPi
$ sudo ./build
$ cd ..
$ swig2.0 -python wiringpi.i
$ sudo python setup.py install
```

neuralassembly, 2017-03-29:
I found that the stall of Rapberry Pi without "sudo" is caused by the official wiringPi, not by the WiringPi-Python. 

When using PWM without "sudo", it stalls, but it works fine with "sudo".
The usual digitalWrite works fine with or without "sudo".

neuralassembly, 2017-03-29:
I added some fixes to the WiringPi-Python for kernel 4.9,
which can be installed as follows.

You have to use "sudo" when you use PWM, if not Rapspberry Pi stalls.
When you do not use PWM, "sudo" would not be required.

```
$ git clone --recursive https://github.com/neuralpi/WiringPi-Python.git
$ cd WiringPi-Python/WiringPi
$ sudo ./build
$ cd ..
$ swig2.0 -python wiringpi.i
$ sudo python setup.py install
$ sudo python3 setup.py install
```

neuralassembly, 2017-03-29:
@Gadgetoid 
Can I make a pull request on the fix of WiringPi-Python for kernel 4.9?
It utilizes the latest wiringPi.

[https://github.com/neuralpi/WiringPi-Python/commit/91b71d701f7b1b56d7e25a7d21a351e104abfb8f](https://github.com/neuralpi/WiringPi-Python/commit/91b71d701f7b1b56d7e25a7d21a351e104abfb8f)

manorius, 2017-05-29:
This totally worked for me! Thanks neuralassembly!

hhenryy, 2017-06-08:
Thanks NeuralAssembly! It Works!!

ipachip, 2017-06-08:
Yep, I've been running with this PR for a while now.  No problems so far.  No idea why it's taking so long to get merged.  Much appreciated @neuralassembly

dehne, 2017-06-15:
Yes, @neuralassembly, much appreciated. Saved me endless hours trying to figure this out.

neuralassembly, 2017-06-25:
I put another comment.
By applying a patch before building WiringPi as shown below,
Wiring-Python will exit with a warning when the hardware PWM is used without "sudo".
Without this patch, Raspberry Pi will stall when the hardware PWM is used without "sudo",
so I recommend you to apply it.

```
$ git clone --recursive https://github.com/neuralpi/WiringPi-Python.git
$ cd WiringPi-Python/WiringPi
$ wget https://raw.githubusercontent.com/neuralassembly/raspi/master/wp-pwm-warning.patch
$ patch -p2 -i wp-pwm-warning.patch
$ sudo ./build
$ cd ..
$ swig2.0 -python wiringpi.i
$ sudo python setup.py install
```

douyining, 2017-07-28:
非常感谢、解决了困扰已久的问题

houhlin, 2017-08-01:
终于解决了，感谢

liffiton, 2017-08-13:
( 👋 Hi, I'm a new maintainer for this repo.)

@neuralassembly, thanks very much for your fix for this.  It's now been merged (after a long wait, yes), and we've pushed a fixed package to pypi (v2.44.3) that should work with that latest Raspbian kernel.  `pip install -U wiringpi` should get everything working now.

The sudo warning patch looks useful, but if I'm not mistaken that makes most sense as a patch for WiringPi upstream.  I'd encourage you to submit it there if you haven't already.

closed 2017-08-13 by liffiton