WiringPi/WiringPi-Python #49: Unable to determine hardware version. I see: Hardware : BCM2835

ifree92, 2017-04-27:
Doesn't work on Raspberry Pi 3 Model B.
(C++ library works as well, and gpio utility works as well too).
But python...

```
$ python3
Python 3.4.2 (default, Oct 19 2014, 13:31:11)
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import wiringpi
>>> wiringpi.wiringPiSetup
wiringpi.wiringPiSetup(      wiringpi.wiringPiSetupGpio(  wiringpi.wiringPiSetupPhys(  wiringpi.wiringPiSetupSys(
>>> wiringpi.wiringPiSetup()
Unable to determine hardware version. I see: Hardware   : BCM2835
,
 - expecting BCM2708 or BCM2709.
If this is a genuine Raspberry Pi then please report this
to projects@drogon.net. If this is not a Raspberry Pi then you
are on your own as wiringPi is designed to support the
Raspberry Pi ONLY.

$ gpio -v
gpio version: 2.44
Copyright (c) 2012-2017 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 3, Revision: 02, Memory: 1024MB, Maker: Embest
  * Device tree is enabled.
  *--> Raspberry Pi 3 Model B Rev 1.2
  * This Raspberry Pi supports user-level GPIO access.
```

imcort, 2017-04-30:
I have the same problem, 
RPi3
Linux version 4.9.24-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #993 SMP Wed Apr 26 18:01:23 BST 2017

NuAoA, 2017-05-04:
Same Issue on a raspberry pi zero

sameer, 2017-05-05:
The WiringPi subrepo needs to be updated to v2.44. The `gpio -v` returns the one you have installed. When you `pip install wiringpi`, it comes bundled with v2.32.

ameenmohamed, 2017-07-09:
same issue tries pip install 
clone and build from git nothing worked yet

IgorGanapolsky, 2017-07-11:
I am having this issue when using Azure IoT.  Running their node.js script produces this error:
> Unable to determine hardware version. I see: Hardware   : BCM2835

nickma82, 2017-07-15:
Thanks to @sameer, [published a fixture](https://github.com/WiringPi/WiringPi-Python/issues/47#issuecomment-310906969) inside that related issue https://github.com/WiringPi/WiringPi-Python/issues/47, which worked for me.

liffiton, 2017-08-13:
This should be fixed with the recent update/release.

closed 2017-08-13 by liffiton

texascanadasean, 2017-12-07:
@IgorGanapolsky  - Did you get a fix for the Azure IoT solution?

IgorGanapolsky, 2017-12-15:
@texascanadasean Yes, I installed the dependency with `npm install wiring-pi`

IgorGanapolsky, 2017-12-18:
@texascanadasean As the author of WiringPi stated to me in an email:
> If you are running Python/PHP/Java/JS/Node, or some language other than
C or RTB BASIC, or some pre-built system such as a TV/Video release or
home automation release,  then sorry - you're screwed. I can't help you.
wiringPi is a C library for C (& C++) and RTB BASIC programs. I do not
support anything else.

jhoughjr, 2019-01-19:
I can inform the author of wiringPi that in 2019 gpio 2.46 command line is broken, no node or java needed.
Just install wiring pi as per instructions here, run gpio and see it somehow doesnt know about the processor in the rpi3.

Funny how following his instructions yields non working software yet its not his problem?
There is no shortage of this issue occurring for anyone using wiringpi in any form. Maybe someone will write a library that is actually maintained and supported. My Pi is a genuine RPI 3 B+.

liffiton, 2019-01-22:
For what it's worth, wiringpi and gpio 2.46 *do* work on an RPi 3 B+.

I've just run this on an RPi 3 B+ with the latest Raspbian and wiringpi installed with `sudo apt install wiringpi`:

```
$ gpio -v
gpio version: 2.46
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 3+, Revision: 03, Memory: 1024MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 3 Model B Plus Rev 1.3
  * This Raspberry Pi supports user-level GPIO access.
```