WiringPi/WiringPi-Python #50: Enable mcp23017 interrupt pin

sky7176, 2017-04-30:
Good morning,
I'm using this library to communicate with an MCP23017.
I would need to enable the interupt on port B configured as input.

I did not understand how to do it and if it can be done using the library.
I searched in the repository but did not find anything.

Now I've walked around the obstacle this way:
Import smbus
Bus = smbus.SMBus (1)
DEVICE_ADDRESS = 0x20
Bus.write_byte_data (DEVICE_ADDRESS, 0x05, 0xFF)

Thanks for your help.
Giuseppe

liffiton, 2018-07-06:
Late, I know, but for anyone with a similar issue:  I don't think wiringPi provides a direct way to do this.  You should be able to use wiringPi's I2C functions to configure the interrupt without having to also import `smbus`, though:

```python
DEVICE_ADDRESS = 0x20
bus = wiringpi.wiringPiI2CSetup(DEVICE_ADDRESS)
wiringpi.wiringPiI2CWriteReg8(bus, 0x05, 0xff)
```

jabss, 2018-08-09:
I'm afraid the wiringPiI2CRead/WriteReg8 functions seems to be broken, so the example above doesn't work...

`pi@raspberrypi:~ $ python
Python 2.7.13 (default, Nov 24 2017, 17:33:09)
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import wiringpi
>>> wiringpi.wiringPiSetup()
0
>>> bus=wiringpi.mcp23017Setup(100, 0x25)  # Need to specify the base, the example above doesn't work 
>>> wiringpi.wiringPiI2CReadReg8(bus, 0x0A)
-1
>>> wiringpi.wiringPiI2CWriteReg8(bus, 0x0A, 0x76)
-1
>>>
`
Other example:

`pi@raspberrypi:~ $ python
Python 2.7.13 (default, Nov 24 2017, 17:33:09)
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import wiringpi
>>> import smbus
>>> bus = smbus.SMBus(1)
>>> wiringpi.wiringPiSetupGpio()
0
>>> wiringpi.digitalRead(17)
1
>>> wiringpi.mcp23017Setup(100, 0x25)
1
>>> wiringpi.digitalRead(100)
0
>>> wiringpi.digitalRead(101)
0
>>> wiringpi.digitalRead(108)
1
>>> wiringpi.digitalRead(107)
0
>>> wiringpi.wiringPiI2CWriteReg8(0x25, 0x0A, 0x74)
-1
>>> wiringpi.wiringPiI2CReadReg8(0x25, 0x0A)
-1
>>> bus.write_byte_data(0x25, 0x0A, 0x74)
>>> wiringpi.wiringPiI2CReadReg8(0x25, 0x0A)
-1
`

I've also noticed that whenever starting up MCP23017, the IOCON register gets updated to 0x20, meaning: 

bit7 - 0 - BANK - Using bank 0
bit6 - 0 - MIRROR - No mirror between INT pins
bit5 - 1 - SEQOP - Sequencial address operation disabled
bit4 - 0 - DISSLW - Slew rate in SDA enabled
bit3 - 0 - HAEN - SPI addresing disabled
bit2 - 0 - ODR - INT Open drain disabled
bit1 - 0 - INTPOL - INT polarity is active-low
bit0 - 0 - Not used

As per the examples above, changing it with wiringpi returns error -1.
I could only find two workarounds:
 - With SMBUS: ex:  bus.write_byte_data(0x23, 0x05, 0xFF) 
 - With OS.SYSTEM: ex: os.system("i2cset -y 1 0x23 0x05 0xFF")

So, any plans to fix the wiringPiI2CRead/WriteReg8 functions ?

Thanks

jabss, 2018-08-10:
Hello,

@liffiton You are right! I was mixing things. 
I beleive Gordon have made a great work wit wiringPI, but I've always had difficulties understanding his documentation. I admit my level of knowledge is very basic, lime may others starting with RPI.
For someone that does't even grasp the concept of what a file descriptor is, reaching the understanding that "int fd" is defining a **f**ile **d**escriptor is a quite a step. I guess a programmers would know, but amateurs like myself....

Anyway, the example below shows how it works. Thanks for your reply.

`pi@raspberrypi:~ $ python
Python 2.7.13 (default, Nov 24 2017, 17:33:09)
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import wiringpi
bus2 = wiringpi.wiringPiI2CSetup(0x25)
bus3 = wiringpi.mcp23017Setup(100, 0x25)
wiringpi.wiringPiSetupGpio()
>>> bus2 = wiringpi.wiringPiI2CSetup(0x25)
print "read 0x0a with WIRPI: ", wiringpi.wiringPiI2CReadReg8(bus2, 0x0A)
print "read 0x05 with WIRPI: ", wiringpi.wiringPiI2CReadReg8(bus2, 0x05)
>>> bus3 = wiringpi.mcp23017Setup(100, 0x25)
print "read GPA7 (104) with WIRPI: ",wiringpi.digitalRead(104)
print "read 0x05 with WIRPI: ", wiringpi.wiringPiI2CReadReg8(bus2, 0x05)>>> wiringpi.wiringPiSetupGpio()
0
>>> print "read 0x0a with WIRPI: ", wiringpi.wiringPiI2CReadReg8(bus2, 0x0A)
read 0x0a with WIRPI:  32
>>> print "read 0x05 with WIRPI: ", wiringpi.wiringPiI2CReadReg8(bus2, 0x05)
read 0x05 with WIRPI:  255
>>> wiringpi.pinMode(104, wiringpi.OUTPUT)         # sets GPA7 to output
>>> print "read GPA7 (104) with WIRPI: ",wiringpi.digitalRead(104)
read GPA7 (104) with WIRPI:  0
>>> wiringpi.digitalWrite(104, 1)
>>> print "read GPA7 (104) with WIRPI: ",wiringpi.digitalRead(104)
read GPA7 (104) with WIRPI:  1
>>> print "read 0x05 with WIRPI: ", wiringpi.wiringPiI2CReadReg8(bus2, 0x05)
read 0x05 with WIRPI:  255
>>>
`

liffiton, 2018-08-10:
Yes, I agree the documentation is not as helpful as it could be, especially for beginners.  Glad you got it worked out, though.

wryan67, 2019-09-27:
I have a library that is focused on using the mcp23017 interrupt pins to write event driven code.  Check it out on github:  https://github.com/wryan67/MCP23x17_RPi

closed 2023-10-31 by Gadgetoid