WiringPi/WiringPi-Python #53: SW PWM works only on GPIO18

arvindpdmn, 2017-06-14:
SW PWM is supposed to work on any GPIO pin but I find that it works only on GPIO18. Below is my code:

    import time
    import sys
    from itertools import chain
    import wiringpi

    def sw_pwm(pin, n=1000):
        wiringpi.pinMode(pin, wiringpi.SOFT_PWM_OUTPUT)
        wiringpi.softPwmCreate(pin, 0, 100)
        for _ in range(n):
            for dc in chain(range(0, 101, 5), range(100, -1, -5)):
                wiringpi.softPwmWrite(pin, dc)
                time.sleep(0.05)

    if __name__ == '__main__':
        LedPin = 18
        wiringpi.wiringPiSetupGpio()

        try:
            sw_pwm(LedPin)
        except KeyboardInterrupt:
            wiringpi.pinMode(LedPin, 0)

Gadgetoid, 2017-08-16:
Are you still having this issue? What are you using to verify that soft PWM isn't working on other pins?

A quick test with an LED (actually the backlight of an LCD) hooked up to BCM 19 and using the following code gets it working on BCM 19 for me:

```
wiringpi.wiringPiSetupGpio()
wiringpi.softPwmCreate(19, 0, 100)
wiringpi.softPwmWrite(19, 50)
```

I didn't need `wiringpi.pinMode(19, wiringpi.SOFT_PWM_OUTPUT)`, although using this doesn't seem to affect my code.

liffiton, 2018-07-02:
With no further details and no one reproducing the bug, I'm closing this to clean up a little.

closed 2018-07-02 by liffiton