WiringPi/WiringPi-Python #55: why doesn't the python-version wiringPi have SPI interfaces?

nuaadot, 2017-06-26:
First , I appreciate your great contribution. Here is my question:
IN the description, we can see : 
description = """A python interface to WiringPi 2.0 library which allows for  easily interfacing with the GPIO pins of the Raspberry Pi. Also supports i2c and SPI""",
And , I also remember in the C-verison wiringpi, we can use the SPI as follows : 
int wiringPiSPISetup (int channel, int speed) ;
int wiringPiSPIDataRW (int channel, unsigned char *data, int len) ;

But， I can't find the similiar function / interface in the python-version
please show me some guidance, Thank you very much!

liffiton, 2017-08-13:
Hi, I'm afraid I can't test them (no SPI hardware), but those functions are present in the module:
```
>>> import wiringpi
>>> wiringpi.wiringPiSPISetup
<built-in function wiringPiSPISetup>
>>> wiringpi.wiringPiSPIDataRW
<built-in function wiringPiSPIDataRW>
```
`wiringPiSPIGetFd` and `wiringPiSPISetupMode` are present, too.  [[source](https://github.com/WiringPi/WiringPi-Python/blob/8a3a64c475687c2ff432d686e437f935e69bda6c/bindings.i#L56)]

Does that help?

closed 2017-08-16 by Gadgetoid

nuaadot, 2017-08-17:
Yes,it does help.Thank you for your patience！