WiringPi/WiringPi-Python #61: Pi Zero File System goes into read only mode when using PWM Functions

ejdigby, 2017-10-04:
When controlling a servo motor using PWM functions inside python3 on a Pi Zero the file system goes into read-only mode and doesn't come out unless you reboot the system.

Commands through the gpio command do work but through python and the wiringpi module doesn't.

liffiton, 2017-10-04:
Thanks for the report.  I'm afraid this one will be difficult to track down, in part because I don't have a Pi Zero to test with.  The most I can offer right now is to try to help you narrow down the cause, and I'm not sure that will get us to a solution.

But if you are willing, please try to produce a minimal example program that exhibits the issue.  Specifically, what is the shortest bit of code that causes the problem, and can it be demonstrated even with nothing external attached to the Pi?

ejdigby, 2017-10-10:
The code I was running was from an [adafruit tutorial](https://learn.adafruit.com/adafruits-raspberry-pi-lesson-8-using-a-servo-motor/overview).

It runs through the program, nothing happens on the output and when you halt the program the filesystem changes into read only mode.

```python
# Servo Control
import time
import wiringpi

# use 'GPIO naming'
wiringpi.wiringPiSetupGpio()

# set #18 to be a PWM output
wiringpi.pinMode(18, wiringpi.GPIO.PWM_OUTPUT)

# set the PWM mode to milliseconds stype
wiringpi.pwmSetMode(wiringpi.GPIO.PWM_MODE_MS)

# divide down clock
wiringpi.pwmSetClock(192)
wiringpi.pwmSetRange(2000)

delay_period = 0.01

while True:
        for pulse in range(50, 250, 1):
                wiringpi.pwmWrite(18, pulse)
                time.sleep(delay_period)
        for pulse in range(250, 50, -1):
                wiringpi.pwmWrite(18, pulse)
                time.sleep(delay_period)
```

ipachip, 2017-10-10:
Do you get any sort of problems if you have a counter instead of while True: ?
I have an rpi 3b and was running a simple script to move a servo with pwm.  It gave me an error (which I don't remember off the top of my head) that indicated something had been unmounted.  
The common threads being pwm and something to do with mounting.  
I updated my system and it seemed to go away but I got the same error once when I was running 
`watch gpio readall`
in one window and my little test script in another.  I'll try to post more details this evening
Chip

Gadgetoid, 2017-12-19:
This may be related to a similar issue from rpi_ws281x where the choice of DMA channel caused an issue: https://github.com/jgarff/rpi_ws281x/issues/208

liffiton, 2017-12-20:
FWIW, I tested the Servo/PWM code from the comment above on a Pi 2 Model B, and there were no errors (filesystem remained RW).  So it doesn't appear to be a general problem with the software.

santtu, 2018-01-08:
I have the very same issue with Raspberry Pi 3 Model B. With version 2.44.4 setting up pin 18 to PWM mode will cause the kernel to immediately reboot the root filesystem read-only:

```
$ python3
Python 3.5.3 (default, Jan 19 2017, 14:11:04)
[GCC 6.3.0 20170124] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import wiringpi
>>> wiringpi.wiringPiSetupGpio()
0
>>> wiringpi.pinMode(18, wiringpi.GPIO.PWM_OUTPUT)
```

When the `pinMode` is issued, the following kernel error occurs:

[image]

I can use the `gpio` command without problems (e.g. `gpio -g mode 18 pwm`).

liffiton, 2018-01-21:
Thanks for the replication and the clear report, @santtu.  On a Pi 2 Model B, your code works with no errors for me when run as root, but when run as a normal user, executing the pinMode line does crash the Pi.  I'm logging in remotely, though, so I can't see whether it's the exact same error.  If you get a chance, could you let me know if the error occurs for you when running python3 with root privileges?

codesinpython, 2018-01-25:
I am running into the same problem using the same examples as provided above but on a different Pi model (noted below). The error only occurs for non-root users, executing it with sudo and everything goes smoothly without error. Could there be an issue with setuid?

Stepping through the code the error occurs at line when setting the pwm range, "wiringpi.pwmSetRange(2000)"

To reiterate, if I run the python interpreter with sudo it runs normally without remounting the filesys in read-only

System Spec:
Raspberry Pi 3 Model B v1.2
Raspbian GNU/Linux 8 (jessie)
python3.4.2
The user is part of the gpio group

# to confirm this is the code I am using
import wiringpi
wiringpi.wiringPiSetupGpio()
wiringpi.pinMode(18, wiringpi.GPIO.PWM_OUTPUT)
wiringpi.pwmSetMode(wiringpi.GPIO.PWM_MODE_MS)
wiringpi.pwmSetClock(192)
wiringpi.pwmSetRange(2000) # error occurs on this line seemingly regardless of value

liffiton, 2018-07-06:
The latest version of wiringpi (2.46) has added [checks](https://git.drogon.net/?p=wiringPi;a=blob;f=wiringPi/wiringPi.c;h=586b148e74b2e9c85e3bc25887b2ed43d8841eb7;hb=HEAD#l681) for certain operations that are not possible when run as a non-root user.  This includes setting a pin's mode to PWM.  With version 2.46.0 of the Python library now:

```
$ python3
Python 3.5.3 (default, Jan 19 2017, 14:11:04)
[GCC 6.3.0 20170124] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import wiringpi
>>> wiringpi.wiringPiSetupGpio()
0
>>> wiringpi.pinMode(18, wiringpi.GPIO.PWM_OUTPUT)
pinMode PWM: Unable to do this when using /dev/gpiomem. Try sudo?
$  # returned to shell -- wiringpi exits whatever program called it -- python3 in this case
```

There's certainly something wrong if the user-level code can crash the entire system, but this update seems to just avoid that issue entirely.  I'll close this now, as the latest version of the library "fixes" it, but feel free to comment if there is still some remaining related problem.

**Fix:** Upgrade to the latest version of wiringpi on PyPI (`pip install -U wiringpi`) and/or don't use PWM without running as root.

closed 2018-07-06 by liffiton