WiringPi/WiringPi-Python #62: Segmentation fault: wiringPiSPIDataRW - wiringpi_wrap.c:1276

isgallagher, 2017-10-17:
Hello,

I am trying to write to and read from the SPI bus via the following command.  It keeps crashing with the below message from gdb.  The exact same function call and code with the C wiringpi library works without issue.

`wiringpi.wiringPiSPIDataRW(0, [6, 0, 0])`

> Starting program: /usr/bin/python3 test.py
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
> 
> Program received signal SIGSEGV, Segmentation fault.
> SWIG_Python_AppendOutput (obj=0x0, result=0x76932918) at wiringpi_wrap.c:1276
> 1276        Py_DECREF(obj);

Thanks,
Ian

liffiton, 2017-10-21:
Thanks for the detailed report.  Can you try passing a bytes object instead of a list?  E.g.:

    wiringpi.wiringPiSPIDataRW(0, bytes([6, 0, 0]))

It looks like that function might only accept bytes objects.  (And maybe strings?  I'm not very familiar with SWIG.)

isgallagher, 2017-10-21:
Hi liffiton,

Yes, that worked to get passed the segmentation fault.  However, now the issue is there is no data being returned.

What I'm actually doing is initializing a buffer variable and passing that.  The function should be passing data back into that same buffer variable, but it is not happening.

E.g.
>>> buf = [6, 0, 0]
>>> wiringpi.wiringPiSPIDataRW(0, bytes(buf))
>>> print buf
[6, 0, 0]

Expected value of buf after the DataRW call would be different values depending on the state of the hardware being polled.

liffiton, 2017-10-21:
I believe the function in Python has been modified to return the data as part of its return value.  The return value looks like it will be a tuple: (length, data).  So try:

    (retlen, retdata) = wiringpi.wiringPiSPIDataRW(0, bytes(buf))

Alternatively, I believe it will work as you expect if you look in exactly the object you passed in (in the code you've written there, the bytes object is a separate object from your `buf` list).  So:

    buf = bytes([6, 0, 0])
    wiringpi.WiringPiSPIDataRW(0, buf)
    print(buf)

isgallagher, 2017-10-21:
I tried both methods you suggested, but the returned variable is always null.  The interesting thing is retlen has an expected length.

>(retlen, retdata) = wiringpi.wiringPiSPIDataRW(0, bytes(buf))
print 'retlen:', retlen, ' retdata:', retdata
retlen: 9  retdata:

liffiton, 2017-10-21:
Hmm.  It's never null in my tests (but do you mean `None` or something else?).  I don't have any SPI devices connected, but the returned buffer is a bytes object containing a series of null bytes (equal in length to the one I passed in).

```python
import wiringpi
wiringpi.wiringPiSPISetup(0, 500000)
buf = bytes([6, 0, 0])
print("Before:", buf)
print("Call returns:", wiringpi.wiringPiSPIDataRW(0, buf))
print("After:", buf)
```
When run, this produces:
```
Before: b'\x06\x00\x00'
Call returns: [3, b'\x00\x00\x00']
After: b'\x00\x00\x00'
```
Are you getting null bytes when you expect non-null bytes, or are you actually getting nothing (i.e., `None`) back?

isgallagher, 2017-10-21:
>Are you getting null bytes when you expect non-null bytes, or are you actually getting nothing (i.e., None) back?

It would actually print blank as if print('').  However, I ran it with python3 and it's working as expected.  Python / python2.7 is returning the blank variable.

Thank you very much for the help, you've been awesome :)

liffiton, 2017-10-21:
No problem.  Glad I could help!

Yeah, I was testing under Python 3.  In Python 2, bytes objects are strings (or something like that?), and so `bytes([6, 0, 0])` is just the string `"[6, 0, 0]"`.  I'm not quite clear on what you were getting back in Python 2, though, because my test under Python 2 does this:

```
Python 2.7.9 (default, Sep 17 2016, 20:26:04)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import wiringpi
>>> wiringpi.wiringPiSPISetup(0, 500000)
3
>>> buf = bytes([6, 0, 0])
>>> print(buf)
[6, 0, 0]
>>> type(buf)
<type 'str'>
>>> wiringpi.wiringPiSPIDataRW(0, buf)
[9, '\x00\x00\x00\x00\x00\x00\x00\x00\x00']
```
So I get back null bytes under Python 2 as well.

Regardless, this looks like a documentation issue at the very least: we should put an SPI example in the README.  I'll get on that.

commit ed4f8cb8 (2017-10-21): Update README.md

Cleaned up and clarified, mostly in the Usage section, and added a section on using SPI (closes #62).
closed 2017-10-21 by liffiton