WiringPi/WiringPi-Python #63: wiringPiISR and segmentation fault

ewok2, 2018-01-23:
Hi
I am using a raspberry (with raspbian) to detect a signal on the Pin 17.
The signal will be sinusaidal.

I write a python script :
`import wiringpi,time
pinCounter = 17
wiringpi.wiringPiSetupGpio()
wiringpi.pinMode(pinCounter, wiringpi.INPUT)

def countImpulse():
 print "top"
 time.sleep(1)

wiringpi.wiringPiISR(pinCounter, `wiringpi.INT_EDGE_RISING,countImpulse)`

but when a activate the signal on the pin 17 I get a segmentation fault from the python cmd line...

liffiton, 2018-02-14:
Thanks for reporting this.  I've reproduced it on my own Pi, running "gpio mode PIN up" and "gpio mode PIN down" on the command line to toggle the pin values without having to connect anything.

I'm afraid this project doesn't have any very active developers right now, so speaking for myself at least, it might be a while before this gets figured out and fixed.

jslay88, 2018-02-14:
Noticing the same on a Pi2 B.

tundratim, 2018-04-22:
Any update on this?  Seeing the same thing on a Pi Zero.

liffiton, 2018-04-28:
Alright, I think I've identified the issue and found a fix.  I'd like to test it a bit more before pushing a new version, though.  Could any of you try to install [the version I've uploaded to test.pypi.org](https://test.pypi.org/project/wiringpi/2.44.5/#files)?  (I'm not sure how often they clean their database on the test instance, but it should be there for a few days at least.)  Download the appropriate `.whl` file based on your version of Python and install it with `pip install FILE.whl`.  If that solves the issue for you, I'll push the new version to PyPI.

jslay88, 2018-04-29:
Will test here in a couple hours on Pi Zero W, 2 B+ and 3 B+.

ewok2, 2018-04-29:
Hy
Wonderful It work for me :-)
I will change my script to use wiringpi again !!!
Many thanks

liffiton, 2018-04-29:
Great!  I've just pushed a new version (2.44.5) to PyPI.

And just for the record: the issue here was caused by me when I moved the `swig` step of building the extension into `setup.py`.  I hadn't realized the `-threads` option needed to be set manually there, and that option was required for the ISR callbacks to function.

closed 2018-04-29 by liffiton