WiringPi/WiringPi-Python #66: Adding support for Rpi 3B + (updating to 2.46)

martinlbb, 2018-07-02:
With the release some month ago of Raspberry Pi 3 model B +, Wiring Pi was updated to 2.46 version (https://git.drogon.net/?p=wiringPi;a=commit;h=8d188fa0e00bb8c6ff6eddd07bf92857e9bd533a).

As WirinPi-Python is still using 2.44 version, there is no support for newer devices.

I manually updated WiringPI to 2.46 version.
Inside WiringPi-Python directory, I did:
```
mv WiringPi WiringPi.2.44
git clone git://git.drogon.net/wiringPi WiringPi
```

Then, I modified the setup.py file to change version from `2.44.6` to `2.46.1`.

Everything worked.
Tested OK on a Rapberry Pi 3 B +:

liffiton, 2018-07-02:
Thanks for pointing that out and testing the new version!  I just uploaded an updated package to PyPI.  I don't have a 3+ to test with myself, so would you mind updating wiringpi with pip (you might need to uninstall the version you created manually first) and let me know here if it works?

martinlbb, 2018-07-07:
I have an old python version (3.2), which is deprectad today.

Pip won't anymore run:
```
pi@pi3 ~/ $ sudo pip3 install wiringpi
/usr/local/lib/python3.2/dist-packages/pip/_vendor/pkg_resources/__init__.py:83: UserWarning: Support for Python 3.0-3.2 has been dropped. Future versions will fail here.
  warnings.warn(msg)
Requirement already satisfied: wiringpi in /usr/local/lib/python3.2/dist-packages
You are using pip version 9.0.1, however version 10.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

However, doing a manual compilation works:
```
pi@pi3 /tmp/ $ git clone --recursive https://github.com/WiringPi/WiringPi-Python.git
pi@pi3 /tmp/ $ cd WiringPi-Python
pi@pi3 /tmp/WiringPi-Python $ sudo python3.2 setup.py install
running install
[........many lines....]
creating /usr/local/lib/python3.2/dist-packages/wiringpi-2.46.0-py3.2-linux-armv7l.egg
Extracting wiringpi-2.46.0-py3.2-linux-armv7l.egg to /usr/local/lib/python3.2/dist-packages
Adding wiringpi 2.46.0 to easy-install.pth file

Installed /usr/local/lib/python3.2/dist-packages/wiringpi-2.46.0-py3.2-linux-armv7l.egg
Processing dependencies for wiringpi==2.46.0
Finished processing dependencies for wiringpi==2.46.0
```

I should be able to get a newer version of raspbian, that let me test Python 3.4 and pip.

liffiton, 2018-07-07:
FWIW, that deprecation notice is just a warning.  The error that prevented installing the library was that you already have a version installed.  `sudo pip3 uninstall wiringpi` should clear that out and let you install the package from PyPI still.

But if the manual installation works on your 3+, that's good enough for me.  Feel free to comment here or open another issue if anything comes up later.  Thanks!

closed 2018-07-07 by liffiton

martinlbb, 2018-07-07:
Thanks for the tip. It worked!

```
pi@pi3 ~ $ sudo pip3 install wiringpi
/usr/local/lib/python3/dist-packages/pip/_vendor/pkg_resources/__init__.py:83: UserWarning: Support for Python 3.0-3.2 has been dropped. Future versions will fail here.
  warnings.warn(msg)
Collecting wiringpi
  Downloading https://files.pythonhosted.org/packages/1b/9b/02f4a84697f53ab1dc2783398956dd252cae71638976553f224a753a15ef/wiringpi-2.46.0.tar.gz (133kB)
    100% |████████████████████████████████| 143kB 718kB/s 
Installing collected packages: wiringpi
  Running setup.py install for wiringpi ... done
Successfully installed wiringpi-2.46.0
```