WiringPi/WiringPi-Python #69: ASUS tinker board

abylikhsanov, 2018-12-23:
Will this library work on ASUS tinker board?

liffiton, 2018-12-23:
It looks like people have used WiringPi on the Tinker Board ([for example](https://tinkerboarding.co.uk/wiki/index.php/GPIO)), so there's a good chance it will.  I haven't tested it, however, nor do I know of any reports of success.

abylikhsanov, 2018-12-28:
I have tried using it on Tinker board and when I try to run example.py I get this:

`wiringPiSetup: mmap (GPIO) failed: Operation not permitted`

Have you faced that before?

abylikhsanov, 2018-12-28:
> Have you tried running it as root or with sudo?

Yes, I did, still the same.

liffiton, 2018-12-29:
I'm afraid I don't know what to suggest, then.  I'm maintaining the python bindings here, but I don't have deep knowledge of the wiringpi library itself, nor do I have a tinker board with which to trace and test this.  I'm sorry.

abylikhsanov, 2019-01-15:
Basically, I just went with ASUS.GPIO and coded everything else.

closed 2019-01-15 by abylikhsanov