WiringPi/WiringPi-Python #70: Future of WiringPi and WiringPi-Python

neuralassembly, 2019-08-22:
Gordon stated that WiringPi is deprecated, and he will release its final version soon.
[wiringPi – deprecated…](http://wiringpi.com/wiringpi-deprecated/?fbclid=IwAR1PXtq_GxRQ_qkEJS0ubd_YojWZQgAMKIGjzlUb1Hg3wybhi11LLVVKoJM)

Will you continue to support  [WiringPi/WiringPi](https://github.com/WiringPi/WiringPi/) and [WiringPi/WiringPi-Python](https://github.com/WiringPi/WiringPi-Python/)?

liffiton, 2019-08-22:
WiringPi-Python has always just been a wrapper around WiringPi, not adding anything to it.  So if WiringPi stops moving, I expect this project will as well.  I plan to still be around as a maintainer of WiringPi-Python, but I probably won't be able to make any substantive improvements.  My guess is that this means it might fail to work with some future hardware revision and/or future kernel version.

I'm really just speaking for myself, though.  As an open source project, others could contribute to it, and I'd be happy to help them where I can.

neuralassembly, 2019-08-22:
Thank you for your reply.

I hope that the repositories [WiringPi/WiringPi](https://github.com/WiringPi/WiringPi/) and [WiringPi/WiringPi-Python](https://github.com/WiringPi/WiringPi-Python/) continue to exist.

I created a patch for [WiringPi/WiringPi-Python](https://github.com/WiringPi/WiringPi-Python/) to support Raspberry Pi 4.
https://github.com/neuralassembly/raspi/blob/master/wp-pi4.patch
This patch is actually for [WiringPi/WiringPi](https://github.com/WiringPi/WiringPi/).

However, I will wait for the final official release that supports Raspberry Pi 4.

liffiton, 2019-08-22:
I'm not an owner of this organization or repositories, so I can't say for sure, but I don't imagine they'll go away.

There may be room to include an unofficial or forked version of WiringPi in this library if it continues development after Gordon stops.  I'm not opposed to it, anyway.  Feel free to open an issue or pull request after we integrate the final official WiringPi here if you have an update we might want to include.  And feel free to open an issue whenever the final official version is released, too, because I am not otherwise paying close attention to that.

davidsean, 2019-11-11:
@neuralassembly , I gave your patch a try and it works as intended. I can now run my python code on a RPi4 which makes me quite happy. Thank you.

neuralassembly, 2019-11-12:
Thank you for your report.
For convenience, I summarize how to use Pi4 patch as follows.

```sudo apt update
sudo apt install python-dev python-setuptools python3-dev python3-setuptools swig
git clone --recursive https://github.com/WiringPi/WiringPi-Python.git
cd WiringPi-Python/WiringPi
wget https://raw.githubusercontent.com/neuralassembly/raspi/master/wp-pi4.patch
patch -p2 -i wp-pi4.patch
sudo ./build
cd ..
swig -python wiringpi.i
sudo python setup.py install
sudo python3 setup.py install
```

liffiton, 2019-12-30:
@neuralassembly, with no new source forthcoming, it seems, I just now merged a pull request from someone else in the WiringPi/WiringPi repo that adds RPi4 support as well.  I don't have an RPi4 to test, so I was just eyeballing it.  It doesn't match yours exactly, so if you see any problems with it, please let me know and I'll fix it.

And I think going forward my plan (unless someone else has better ideas, which would be fine with me) is to continue to add small updates to WiringPi/WiringPi as needed to support new hardware going forward.  I probably won't be writing those updates myself, but I'm happy to merge others' work.  And as long as I have time, I'll keep merging those changes into the Python port here as well.  I welcome any and all help with that.

neuralassembly, 2019-12-30:
@liffiton OK. I will test it soon.

liffiton, 2020-01-01:
With code from @neuralassembly and @steveb, I updated the main WiringPi/WiringPi repository, and I've just now updated this repo and pushed new packages to PyPi.  "pip install" should get a version compatible with RPi4 now.  I really appreciate the help, because that update wasn't something I could do on my own.

I'm happy to keep making small compatibility updates like that when I have time.  Anyone else should feel free to contact the [Github organization owners](https://github.com/orgs/WiringPi/people) to request to be added as a maintainer to make changes, too.  That's what I did a while back when there was a period with nobody actively maintaining the Python port.

closed 2020-01-01 by liffiton