WiringPi/WiringPi-Python #73: installation failure on archlinux: compilation error

etienne1911, 2020-09-19:
Hello,

Trying to install wiringPi on RaspberryPI Zero fails with the following compilation error on ArchLinuxArm :

`gcc -pthread -shared -Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now -fno-semantic-interposition -Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now build/temp.linux-armv6l-3.8/WiringPi/devLib/scrollPhat.o build/temp.linux-armv6l-3.8/WiringPi/devLib/piNes.o build/temp.linux-armv6l-3.8/WiringPi/devLib/piGlow.o build/temp.linux-armv6l-3.8/WiringPi/devLib/piFace.o build/temp.linux-armv6l-3.8/WiringPi/devLib/maxdetect.o build/temp.linux-armv6l-3.8/WiringPi/devLib/lcd128x64.o build/temp.linux-armv6l-3.8/WiringPi/devLib/lcd.o build/temp.linux-armv6l-3.8/WiringPi/devLib/gertboard.o build/temp.linux-armv6l-3.8/WiringPi/devLib/ds1302.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/wpiExtensions.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/wiringShift.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/wiringSerial.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/wiringPiSPI.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/wiringPiI2C.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/wiringPi.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/sr595.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/softTone.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/softServo.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/softPwm.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/sn3218.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/rht03.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/pseudoPins.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/piThread.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/piHiPri.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/pcf8591.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/pcf8574.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/mcp4802.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/mcp3422.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/mcp3004.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/mcp3002.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/mcp23s17.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/mcp23s08.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/mcp23017.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/mcp23016.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/mcp23008.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/max5322.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/max31855.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/htu21d.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/ds18b20.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/drcSerial.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/drcNet.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/bmp180.o build/temp.linux-armv6l-3.8/WiringPi/wiringPi/ads1115.o build/temp.linux-armv6l-3.8/wiringpi_wrap.o -L/usr/lib -o build/lib.linux-armv6l-3.8/_wiringpi.cpython-38-arm-linux-gnueabihf.so -lcrypt -lrt
    /usr/bin/ld: build/temp.linux-armv6l-3.8/WiringPi/wiringPi/drcNet.o:/tmp/pip-install-nfcg0jo_/wiringpi/WiringPi/wiringPi/../wiringPiD/drcNetCmd.h:43: multiple definition of `comDat'; build/temp.linux-armv6l-3.8/WiringPi/wiringPi/wpiExtensions.o:/tmp/pip-install-nfcg0jo_/wiringpi/WiringPi/wiringPi/../wiringPiD/drcNetCmd.h:43: first defined here
    collect2: error: ld returned 1 exit status
    error: command 'gcc' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/pizerow/.local/share/virtualenvs/motor_mc33926-U0W_KJio/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-nfcg0jo_/wiringpi/setup.py'"'"'; __file__='"'"'/tmp/pip-install-nfcg0jo_/wiringpi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-0gtnf5ql/install-record.txt --single-version-externally-managed --compile --install-headers /home/pizerow/.local/share/virtualenvs/motor_mc33926-U0W_KJio/include/site/python3.8/wiringpi Check the logs for full command output.
`
Looking at the issue I may have found the reason (see on this [thread](https://www.linuxquestions.org/questions/programming-9/multiple-definition-errors-when-linking-on-arch-using-gcc-10-1-0-a-4175675444/))
where it seems the error would be due to a newer version of gcc (.

Could you confirm this is linked and do you have any idea of a fix?
Thanks

etienne1911, 2020-09-19:
After digging a bit I saw there was already a fix comming in the WiringPi repository

https://github.com/WiringPi/WiringPi/pull/76/commits/c69c7d60058e04f1e170723c2a9562ea095d1c79 

Could you please approve it so that arch user can install it?
Sorry for opening this issue

liffiton, 2020-09-23:
Thanks for reporting that.  It's worth keeping this open until a fix is applied *and* pulled into this library and released.

Link to the relevant pull-request: WiringPi/WiringPi#76

steve51516, 2021-02-20:
Was this resolved? I'm having the same problem (or similar) on Arch linux Raspberry pi 3B+
gcc exits with code 1
```bash
[steven@alarmpi Adafruit_BluefruitLE-0.9.10]$ pip install wiringpi
Defaulting to user installation because normal site-packages is not writeable
Collecting wiringpi
  Downloading wiringpi-2.60.0.tar.gz (130 kB)
     |████████████████████████████████| 130 kB 636 kB/s 
Using legacy 'setup.py install' for wiringpi, since package 'wheel' is not installed.
Installing collected packages: wiringpi
    Running setup.py install for wiringpi ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-lgr0bq9t/wiringpi_6987d2d4a5fd4a9fb5ee4a6cabf6e9a6/setup.py'"'"'; __file__='"'"'/tmp/pip-install-lgr0bq9t/wiringpi_6987d2d4a5fd4a9fb5ee4a6cabf6e9a6/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-99eijyvi/install-record.txt --single-version-externally-managed --user --prefix= --compile --install-headers /home/steven/.local/include/python3.9/wiringpi
         cwd: /tmp/pip-install-lgr0bq9t/wiringpi_6987d2d4a5fd4a9fb5ee4a6cabf6e9a6/
    Complete output (103 lines):
    running install
    running build
    running build_py
    running build_ext
    building '_wiringpi' extension
    swigging wiringpi.i to wiringpi_wrap.c
    swig -python -threads -o wiringpi_wrap.c wiringpi.i
    creating build
    creating build/temp.linux-armv7l-3.9
    creating build/temp.linux-armv7l-3.9/WiringPi
    creating build/temp.linux-armv7l-3.9/WiringPi/devLib
    creating build/temp.linux-armv7l-3.9/WiringPi/wiringPi
    gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -fno-semantic-interposition -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -fPIC -IWiringPi/wiringPi -IWiringPi/devLib -I/usr/include/python3.9 -c WiringPi/devLib/ds1302.c -o build/temp.linux-armv7l-3.9/WiringPi/devLib/ds1302.o
    gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -fno-semantic-interposition -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -fPIC -IWiringPi/wiringPi -IWiringPi/devLib -I/usr/include/python3.9 -c WiringPi/devLib/gertboard.c -o build/temp.linux-armv7l-3.9/WiringPi/devLib/gertboard.o
    gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -fno-semantic-interposition -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -fPIC -IWiringPi/wiringPi -IWiringPi/devLib -I/usr/include/python3.9 -c WiringPi/devLib/lcd.c -o build/temp.linux-armv7l-3.9/WiringPi/devLib/lcd.o
    gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -fno-semantic-interposition -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -fPIC -IWiringPi/wiringPi -IWiringPi/devLib -I/usr/include/python3.9 -c WiringPi/devLib/lcd128x64.c -o build/temp.linux-armv7l-3.9/WiringPi/devLib/lcd128x64.o
    gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -fno-semantic-interposition -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -fPIC -IWiringPi/wiringPi -IWiringPi/devLib -I/usr/include/python3.9 -c WiringPi/devLib/maxdetect.c -o build/temp.linux-armv7l-3.9/WiringPi/devLib/maxdetect.o
    gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -fno-semantic-interposition -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -fPIC -IWiringPi/wiringPi -IWiringPi/devLib -I/usr/include/python3.9 -c WiringPi/devLib/piFace.c -o build/temp.linux-armv7l-3.9/WiringPi/devLib/piFace.o
    gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -fno-semantic-interposition -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -fPIC -IWiringPi/wiringPi -IWiringPi/devLib -I/usr/include/python3.9 -c WiringPi/devLib/piGlow.c -o build/temp.linux-armv7l-3.9/WiringPi/devLib/piGlow.o
    gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -fno-semantic-interposition -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O3 -pipe -fno-plt -fPIC -IWiringPi/wiringPi -IWiringPi/devLib -I/usr/include/python3.9 -c WiringPi/devLib/piNes.c -o build/temp.linux-armv7l-3.9/WiringPi/devLib/ [... cut ...]
ERROR: Command errored out with exit status 1: /usr/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-lgr0bq9t/wiringpi_6987d2d4a5fd4a9fb5ee4a6cabf6e9a6/setup.py'"'"'; __file__='"'"'/tmp/pip-install-lgr0bq9t/wiringpi_6987d2d4a5fd4a9fb5ee4a6cabf6e9a6/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-99eijyvi/install-record.txt --single-version-externally-managed --user --prefix= --compile --install-headers /home/steven/.local/include/python3.9/wiringpi Check the logs for full command output.
```

liffiton, 2021-02-20:
Oddly enough, I was just looking back at this today.  I did merge the fix in the main WiringPi repository a while, but I never came back to bring it in to this Python library.  So I still need to do that.  I'll work on it now.

For now, though, my apologies for not getting it done sooner.

liffiton, 2021-02-20:
@steve51516, just as a potential workaround and for my own education, could you try installing the `wheel` package with `pip install wheel`, then trying to install `wiringpi` again?  It may work, as the wheel is precompiled.

liffiton, 2021-02-20:
Oh, no, on Arch you probably have Python 3.8, for which there isn't a wheel on PyPI anyway.  So that won't work in any case, I think.  :slightly_smiling_face:

liffiton, 2021-02-20:
Alright!  The latest WiringPi code (with the fix for Arch and other more recent GCC installs) is pulled in with commit 93a6e40, and a new version (2.60.1) is published on PyPI.  It should install correctly now.  If not, please reopen this issue if it's the same error still, or open a new one if something else goes wrong.

closed 2021-02-20 by liffiton

steve51516, 2021-02-20:
@liffiton It works! Thank you for the quick fix.