WiringPi/WiringPi-Python #77: No "Revision" line ERROR

raisondet, 2021-04-12:


raisondet, 2021-04-12:
Hi~
I used the VIM3(https://www.khadas.com/vim3) board that is installed Ubuntu.
I am trying to use wiringpi-python on the VIM3 board.
When I run below code, some error is occured.
===> No "Revision" line

[image]

this is /proc/cpuinfo
[image]

What should I do to fix this bug?

liffiton, 2021-04-12:
That looks like the same issue as WiringPi/WiringPi#107.  It's an incompatibility between WiringPi and your OS, and it's not something this Python library can fix itself.  And I'm afraid no fix is immediately forthcoming in WiringPi.

Beyond that, though, I'm not sure WiringPi will be compatible with non-RaspberryPi boards, so even if this issue were fixed, you may still not be able to use WiringPi on your board.

Gadgetoid, 2021-04-13:
> I'm not sure WiringPi will be compatible with non-RaspberryPi boards

I mean, clue's in the name :laughing:

WiringPi (and most other Raspberry Pi IO libraries to be fair) bitbangs the Raspberry Pi's GPIO directly and is fundamentally incompatible with any other boards. For cross-platform GPIO use libgpiod. For fast, bitbang GPIO access on <insert platform here> you'll need to write your own bitbang implementation.

closed 2021-04-13 by Gadgetoid